esenapaj/Marlin #146: [REQUEST] Include pins definitions for DUE3DOM and DUE3DOM MINI

GarrethX, 2016-07-05:
Hello,

I'm sorry if this is not appropriate place to put this request, however I don't know how to contact you the other way and I don't know how to make my own pull requests to do it myself.

On polish reprap discussion board, administrator designed in collaboration with other users and manufactured two expansion boards for arduino due (link: www.due3dom.pl ).

We've got two pins files created and tested for these boards:
Due3dom: https://dl.dropboxusercontent.com/u/25040030/3d%20printer/due3dom/pins_DUE3DOM.h
Due3dom MINI: https://dl.dropboxusercontent.com/u/25040030/3d%20printer/due3dom/pins_DUE3DOM_MINI.h

It would be great if these pin files could be added in your repository for marlin due, so that future firmware updates would be easier for us. 

Thank you in advance!

esenapaj, 2016-07-10:
I've executed your order just now.
If any problems exist, please report.
If it's no problem, please close this topic.

GarrethX, 2016-07-10:
Thank you very much, you're the best:)

Closing now

closed 2016-07-10 by GarrethX