esenapaj/Marlin #156: problem with drv8825 stepper

dp90pusika, 2016-08-10:
I have drv8825 stepper drivers and they not able to move the motors properly.
They move but slower than they should, and also rally noisy like the skip a lot of steps.
I attach a youtube video so you can se what I mean exactly.
https://youtu.be/ROyEuKbiON8
I have a RADDS control board with an arduino due. 

After dooing some reserch I think ist's something to do with pulse lenght. Can you direct me to the part of the script I should look at and modify?

esenapaj, 2016-08-10:
It's likely these...
RADDS V1.5 + DRV8825 - Z axis not moving - Google グループ
https://groups.google.com/forum/#!topic/wanhao-printer-3d/qTsZL4DP1sE

Repetier 0.91 rev 7, h bot uneven stepping
http://forums.reprap.org/read.php?267,356502

PLA extreme bridging torture test - YouTube
https://www.youtube.com/watch?v=bEWcfoamTQo

>Can you direct me to the part of the script I should look at and modify?

Maybe here...?
If you inclement delay count (+1U ~ 4U), problem is solved possibly.
https://github.com/esenapaj/Marlin/blob/Duo/Marlin/stepper.cpp#L1172

dp90pusika, 2016-08-10:
thanks for the reply, 
the first 3 link is for repetier.
Repetier has a  #define STEPPER_HIGH_DELAY  which fixes the problem, but repetier has a crappy dual x carriage support. 
I would like to use marlin.
I tried to modyfy the delay you suggested up to (10u) but no effect.
After that I tried doubbleing all the delays I could find , but still no effect.

I have a marlin working variant of marlin: https://github.com/Wurstnase/Marlin4Due
this one drives the motors correctly but misses a lot of functionality which your Marlin has.
Can you posibbly look at the two HAL files to identify the mayn diferenc?
I tried it my self but they are so diferent I can't get a sense of it.

esenapaj, 2016-08-10:
If no problem with Marlin4Due, cause of this problem isn't delay in https://github.com/esenapaj/Marlin/blob/Duo/Marlin/stepper.cpp#L1172.
Because [delay time of this in Marlin4Due](https://github.com/Wurstnase/Marlin4Due/blob/master/Marlin/stepper.cpp#L1094) as same as mine,
and [content of _delay_us()](https://github.com/Wurstnase/Marlin4Due/blob/master/Marlin/HAL.h#L80-L88) is also same as [mine](https://github.com/esenapaj/Marlin/blob/Duo/Marlin/HAL.h#L116L125).

Then, probably cause is delay in multi (double / quad) stepping mode.
In case of Malin4Due, it doesn't use multi stepping when `ENABLE_HIGH_SPEED_STEPPING` isn't enabled.
In case of my Marlin Duo, it use multi stepping automatically as official Marlin. 

 | Marlin4Due | Marlin4Due | Marlin Duo | official Marlin
--- | --- | --- | --- | ---
 | `ENABLE_HIGH_SPEED_STEPPING` is disabled |  `ENABLE_HIGH_SPEED_STEPPING` is enabled | `ENABLE_HIGH_SPEED_STEPPING` option doesn't exist | `ENABLE_HIGH_SPEED_STEPPING` option doesn't exist
`MAX_STEP_FREQUENCY` | 150000Hz | 500000Hz | 320000Hz | 40000Hz
single stepping | ~ 150000Hz | ~ 100000Hz | ~ 90000Hz | ~ 10000Hz
double stepping | disabled | 100001 ~ 200000Hz | 90001 ~ 180000Hz| 10001 ~ 20000Hz
quad stepping | disabled | 200001Hz ~ 500000Hz | 180001 ~ 320000Hz | 20001 ~ 40000Hz

Maybe your problem is occurring in over 90000Hz.
But Marlin Duo hasn't delay in multi stepping mode like a `DOUBLE_STEP_DELAY` of Repetier-Firmware currently, then I try to add it.
This is a test branch for you
https://github.com/esenapaj/Marlin/tree/Duo_test
, it adds a new option `MULTI_STEPPING_DELAY`, it's similer to `DOUBLE_STEP_DELAY` of Repetier-Firmware.

dp90pusika, 2016-08-10:
Thank you @esenapaj! I will test it right away.
The frequency could be an issue because i have 400steps/mm and generally print ower 100mm/s

dp90pusika, 2016-08-10:
okey, tried MULTI_STEPPING_DELAY 2, 5, 16, 100
and no change at all, sadly :(

esenapaj, 2016-08-11:
Oh, sorry...
But, still another solutions exists.
https://github.com/esenapaj/Marlin/tree/Duo_test2

`STEP_DOUBLER_FREQUENCY`:
A threshold value between single and double stepping.
It's similar to `STEP_DOUBLER_FREQUENCY` of Repetier-Firmware.
Double stepping start from `STEP_DOUBLER_FREQUENCY` + 1, quad stepping start from `STEP_DOUBLER_FREQUENCY` * 2 + 1.
Default value is 900000 (Hz).
To see [here](https://github.com/repetier/Repetier-Firmware/issues/410) and [here](https://www.youtube.com/watch?v=bEWcfoamTQo), it seems that 700000Hz is good for DRV8825? 

`STEPPER_HIGH_DELAY`:
Add additional delay for pulse signal of stepper.
It's similar to `STEPPER_HIGH_DELAY` of Repetier-Firmware.

`MULTI_STEPPING_DELAY`: (same as yesterday's solution)
Add additional delay in double / quad stepping mode of stepper.
It's similar to `DOUBLE_STEP_DELAY` of Repetier-Firmware.

`DISABLE_MULTI_STEPPING`:
It disables double / quad stepping simply.
This is bad solution, but better than the broken stepping...

Anyway, I have to obtain real thing of DRV8825 for solving this problem.
Thus I ordered DRV8825 and RADDS today.
But I think that It take a time for those to be delivered to me...

dp90pusika, 2016-08-11:
Thank you @esenapaj for your dedication!
sadly none of the modifications work
STEPPER_HIGH_DELAY only fork in combination with STEP_DOUBLER_FREQUENCY, I tried these first with multiple values, DELAY:1,2,4,5,10,20,40 and FREQUENCY: 70000, 50000, 20000, 90000, 150000
These had no effect.
Then I tried only the FREQUENCY with the values above, still nothing.

MULTI_STEPPING_DELAY also works only in combination with FREQUENCY so I went through with the frequency walues above and tried DELAYs:1,2,4,5,10,20,40
still no effect.

Also used some random config on all of them, but no result at al.

Then used DISABLE_MULTI_STEPPING and still no change. 
I'm completly baffled on this, i spent two days now comparing the stepper.cpp and the hal.h and other scripts line by line with the working marlin4due and they look the same in eweri detail except for some additional lines. Im lost.

esenapaj, 2016-08-11:
Oh... Um...
By the way, Did you try to clear EEPROM?
```
M502
M500
``` 
M502 loads default settings, M500 saves settings to EEPROM.

dp90pusika, 2016-08-11:
I think it's disabled, but checking now

dp90pusika, 2016-08-11:
`//
// EEPROM
//
// The microcontroller can store settings in the EEPROM, e.g. max velocity...
// M500 - stores parameters in EEPROM
// M501 - reads parameters from EEPROM (if you need reset them after you changed them temporarily).
// M502 - reverts to the default "factory settings".  You still need to store them in EEPROM afterwards if you want to.
//define this to enable EEPROM support
//#define EEPROM_SETTINGS

#if ENABLED(EEPROM_SETTINGS)
  // To disable EEPROM Serial responses and decrease program space by ~1700 byte: comment this out:
  #define EEPROM_CHITCHAT // Please keep turned on if you can.
#endif

//
// Host Keepalive
//
// When enabled Marlin will send a busy status message to the host
// every couple of seconds when it can't accept commands.
//
#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages
#define DEFAULT_KEEPALIVE_INTERVAL 2  // Number of seconds between "busy" messages. Set with M113.
`
Yes definiteli disabled

esenapaj, 2016-08-11:
Yes, your EEPROM on RADDS is disabling.
But then, investigation became much difficult.

EEPROM isn't a cause.
Additional delays are meaningless, Forcing of single stepping is also meaningless. 
32bit part of Marlin Duo is based on Marlin4Due, differences are small as you said.
In the first place, Marlin4Due hasn't special additional delay for slow stepper.
Nevertheless Marlin4Due works fine.

Therefore, I think that we've no choice but to wait until I obtain DRV8825 and RADDS.
But we can try some thing if you're interested.

__1. Check your config__
If you change names of your `Configuration.h` and `Configuration_adv.h` to `Configuration.txt` and `Configuration_adv.txt`
and drop to comment column, you can upload full of configurations to here, then I can research it.
Or, you can copy & paste all the contents of `Configuration.h` and `Configuration_adv.h` to here simply.

__2. Search a "working point"__
This is a simple but much trouble solution for research.
You can see much `<>` on right of [page of commit log](https://github.com/esenapaj/Marlin/commits/Duo) and click a one at random.
Then you are moved to new page, and you can click "Clone or download" - "Download ZIP".
The .zip is a "branch at certain point of time". Videlicet, you can test old Marlin Duo.
If you are lucky, Possibly you can find a "Marlin Duo that it's before broken".
Reiterate, this is much trouble.
Therefore you don't needs to execute this if you aren't interested this.
Even if you don't execute I'll execute this after I obtain DRV8825.

dp90pusika, 2016-08-11:
I'll try to help as much as I can.

[Configuration.txt](https://github.com/esenapaj/Marlin/files/413781/Configuration.txt)
[Configuration_adv.txt](https://github.com/esenapaj/Marlin/files/413782/Configuration_adv.txt)

I will start the search for the working one for you, if we are lucky I find the last working iteration for you, before your DRVs and RADDS arrive, maybe it will be easier for you to work out the solution.

Also, the main reason to update my firmware to Marlin duo is because I use a dual x carriage setup, and I need high speed tool changing for the best result. Marlin4Due starts to to behave weird above 350mm/s travel speed. It travels fine for the most of the distance and a fev mm befor it reaches the coordinate, it slowes down to like 1mm/s and finishes the move like that. It looks like it can't load the next move in to the buffer so it slowes down the movement. 

The second reason is that with Marlin4due in duplicate mode the second extruder motor doesn't move, like it's not enabled, but I can't find the piece of code that copies the E_STEPS to the second extruder. The movement othervise is fine.

So I'm hopeing to solve these two problems with updating to your Firmware.

PPs. I allso have dual z motors and dual z endstop, that could be a problem later down the line. Before i used a RADDS board I managged to use a modified RAMPS1.4 with  MEGA board, and there were problems with the dual z homing. Howewer your firmware doesent seem to have problems at the compiling so it looking good for now, but can't test untill the motors move the axes properly. The 8 bit wersion had problems after RC3 I believe. But this is a problem for later

esenapaj, 2016-08-11:
>Configuration.txt
>Configuration_adv.txt

Hum... It looks like that it's not strange.
It's much the same as default setting.
And I've less knowledge about cartesian, because I've only delta printer.

If I point that I'm concerned, `#define SLOWDOWN` in `Configuration_adv.h`.
This option is [enabled by default on Marlin Duo](https://github.com/esenapaj/Marlin/blob/Duo/Marlin/Configuration_adv.h#L364) and [official Marlin](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration_adv.h#L364),
but this is [disabled by default on Marlin4Due](https://github.com/Wurstnase/Marlin4Due/blob/master/Marlin/Configuration_adv.h#L254).
I know that this option causes stuttering at high-load.

dp90pusika, 2016-08-16:
Hey I finally narrowed it down. it seems on feb 29 you did some changes in the stepper.cpp, and from that on it wasn't working at all till 29 days ago when you did a merge with the Merge remote-tracking branch.
can you take a look at the changes in stepper.cpp on fc31b82  on Feb 29?
working: 
	Latest commit 3465424  on Feb 12
	Latest commit 5e2c66f  on Feb 25 /works fine up to 800mm/s
	Latest commit 7cd3689  on Feb 26 /ok
	Latest commit b6d9387  on Feb 28 /ok

not working: 
	Latest commit fc31b82  on Feb 29 //frosen axes !!!!!!!!the first one wich doesent work!!!!!!!!!
	Latest commit e219f0b  on Feb 29 //frosen axes
	Latest commit 46d5782  on Mar 2  //frosen axes
	Latest commit f2c8a04  on Mar 15 //frosen axes
	Latest commit 23fd574  on Mar 31 //frosen axes
	Latest commit 97a9ea9  on Apr 2  //frosen axes
	Latest commit ce349dc  on Apr 15 //hall.cpp delay_ms compiling error
	Latest commit 15e214a  on May 1  //EEPROM compilation error
	Latest commit b4eca2e  on May 16 //still frosen axes
	Latest commit 687c08b  on Jun 3  //undefined temp settings compillation error
	Latest commit 8c90a39  on Jun 8  //still frosen axes
	Latest commit 04085dd  on Jul 2  //still frosen axes
	Latest commit a6014c1  on Jul 17 //still frosen axes

moves but loosing stepps:
	Latest commit dcafbcb  29 days ago
	Latest commit 28a550c  25 days ago

esenapaj, 2016-08-16:
I'm surprising to your passion. I really appreciate your cooperation.
I'll start analyzing tomorrow.
Apparently, a cause seems that it's one of black box for me.
`STEP_ADD` and `STEP_IF_COUNTER` macros of Due is differing with those of AVR.
It was introduced in commit 384e78acffc58420f42f3557e1030652cc1823f2 (stepper rework),
but I don't understand currently why the changing was needed.

And, I've obtained RADDS and new Due yesterday. I feel that RADDS has good-looking and well thought-out components.
[image]
But DRV8825 is still in transit.
And, I have to improvement LCD support of RADDS in my `pins_RADDS.h`.

dp90pusika, 2016-08-16:
Isn't it beautiful? :D
I also appreciate your help, this is the least I can do since I can't code well enough. 
But I try my best to learn, because I will really need it for what I'm planning.
If this will work I want to use this as a base platform to build a PID loop control for X Y axis. I designed/built this machine, now it's time to learn to program it :D

esenapaj, 2016-08-17:
Would you test `STEPPER_HIGH_DELAY` again?
https://github.com/esenapaj/Marlin/tree/Duo_test_3

dp90pusika, 2016-08-19:
Sorry I was out for a while. 
I tried it and it behaves like the versions from Feb 29 to July 17, the axis won't move at all.
If I try to move an axis the motors get enabled but no movement at all.

esenapaj, 2016-08-19:
I've obtained DRV8825 x 5, and start setting.
But DRV8825 that it was set to z axis always scream loudly, and doesn't move.
another DRV8825 that it was set to e0 doesn't scream, but doesn't move as z axis...
I'm stumbling at first step.

dp90pusika, 2016-08-19:
That's great news, it will be alot easier for you now. I understand even less than you, there shouldn't be any problem with these steppers. More and more manufacturers are using it in their controler boards because it's not that expensive but double the resolution. Is it possibble that so few people are using it that noone had this problem befor?

esenapaj, 2016-11-05:
Sorry for keeping you waiting for such a long time, I've confirmed that my DRV8825 works with [`MINIMUM_STEPPER_PULSE`](https://github.com/esenapaj/Marlin/blob/Duo_test_final/Marlin/Configuration_adv.h#L572) = __2__.
It has worked since over one month ago, but I didn't realized it because I didn't realized that my first all the DRV8825 x 5 broke by wrong vref setting at [first day, first test](https://github.com/esenapaj/Marlin/issues/156#issuecomment-241049349).
I purchased DRV8825 x 5 again, and finally I've confirmed that it's working.
https://github.com/esenapaj/Marlin/tree/Duo_test_final

`MINIMUM_STEPPER_PULSE` = __0__ (default)

XYZ axes and E axis (not advanced algorithm) | E axis (`ADVANCED`) | E axis (`LIN_ADVANCED`)
--- | --- | ---
[image] | [image] | [image]
HIGH pulse width = 1.180μs, LOW pulse width = 1.000μs | HIGH pulse width = 48.00ns, LOW pulse width = 298.0ns | HIGH pulse width = 380.0ns, LOW pulse width = 250.0ns

`MINIMUM_STEPPER_PULSE` = __2__

XYZ axes and E axis (not advanced algorithm) | E axis (`ADVANCED`) | E axis (`LIN_ADVANCED`)
--- | --- | ---
[image] | [image] | [image]
HIGH pulse width = 3.240μs, LOW pulse width = 2.320μs | HIGH pulse width = 2.520μs, LOW pulse width = 2.340μs | HIGH pulse width = 2.560μs, LOW pulse width = 2.500μs

From [datasheet](http://www.tij.co.jp/product/DRV8825/datasheet/specifications), DRV8825 needs 1.9μs HIGH/LOW pulse width at least, and 250KHz (4μs) step frequency(pulse period)is maximum.

minimum requirements specification | DRV8825
--- | --- | ---
HIGH pulse width (μs)  | 1.9
LOW pulse width (μs) | 1.9
Pulse period (μs) | 4.0

dp90pusika, 2016-11-08:
Thank you for your hard work!
I already discovered that it works, and did some testing with it.
Works grate in my printer.
Thank you for fixing this!

Sadly I still can't use it, because I have dual z endstop enabbled, and that part is also busted.
I'm thinking about opening a new issue on the main branch, or do you have the time to take a look at it?

esenapaj, 2016-11-08:
Glad problem of your stepper was solved.

>I have dual z endstop enabbled, and that part is also busted.

About Dual X carriage and Z Homing, It seems that (official) Marlin has long standing issues.

Dual_X_Carriage proper behavior is broken · Issue #4694 · MarlinFirmware/Marlin
Dual_X_Carriage not respecting soft Z max value · Issue #4695 · MarlinFirmware/Marlin

And, now patch for those exist but it isn't merged yet.
Unpark when using Z_SAFE_HOMING with Dual X by thinkyhead · Pull Request #5175 · MarlinFirmware/Marlin

So, I've created a test branch that it include above patch.
Does this solves your z problem?
https://github.com/esenapaj/Marlin/tree/Duo-testes

dp90pusika, 2016-11-09:
It solved the z homing problem, it beautifully works homes to both z endstops.
I mowed around the axis from the control panel and everything seemed to work fine.
Than I started a test print, that also worked fine. 

But then I started a duplication print, and everything went to south. the bed went down rapidly about 150mm than rapidly back up but the motors skipped so it never reached the original position. than the second cariage moved in position and started normaly working in duplication mode.
After an emergency stop, I re connected and started it again, but all the axis went crazy. For a homing command both x axis went toward their end stops and crashed into them for a time, than they did a homing bump, same with the y axes, and the z still dropped down.

I tried to set back the default eeprom, it didn't help.
Then I re flashed the firmware with eeprom disabled, but this weird behavior persisted.

esenapaj, 2016-11-17:
Oh I'm sorry, I forgot to respond.

It looks like that homing problem is solved by above PR #5175, however another problem (on duplication print) manifested.
I think that it's problem of part of original Marlin because it seems unrelated to the 32 bit part.
But I’m not sure how to fix it because I'm not familiar with Dual X carriage.

I propose that you post on the official forum on that problem,
But I'd like to ask you, if you still have MEGA2560 and RAMPS (8bit system),
please you revalidate using the the system and official Marlin before report to the official forum.
Because my 32bit branch is forked and unofficial, therefore it isn't preferable to report the results of verification with forked branches only to the official forum.

esenapaj, 2016-11-18:
I've created a official RCBugFix + PR #5175 (Unpark when using Z_SAFE_HOMING with Dual X) branch.
https://github.com/esenapaj/Marlin/tree/Official-RCBugFix-plus-PR-%235175

thinkyhead, 2016-11-19:
I need to get #5175 completed. I hope I will be able to consult with @AndrewDiehl asap and figure out how to make it all perfect.

dp90pusika, 2016-11-22:
Sorry guys, I needed some time to rig up a test setup because my IDEX printer is close to its final form, I didn't wanted to take it apart.

@esenapaj this official RCBugFix + PR #5175 is just as faulty like the previous version, as far as I can tell.
I made a video of the behavior, here is the link to it: https://youtu.be/-7jUi-At8FM
it is pretty long and boring, so here is a timeline to help explain what is happening:
[Video timestamp.txt](https://github.com/esenapaj/Marlin/files/606107/Video.timestamp.txt)

I also copied the end of the communication log form the control panel, it's in this file:
[last of communication.txt](https://github.com/esenapaj/Marlin/files/606110/last.of.communication.txt)

and this is the g-code I was using, it is a simples dual color calibration:
[calibration cubes g-code.txt](https://github.com/esenapaj/Marlin/files/606114/calibration.cubes.g-code.txt)

What I can tell from these tests, is that the firmware on tool change gets its feed rate and position data from somewhere that is independent from the max feed rate and max position. The max position was set to 500mm on Z than I changed it to 100 and it didn't make ani diference in these long travels.
Allso the toolchange seems to make the carriages move to the home direction only, maybe they don't get their dir pin flipped.
And the last thing, when tool change occurs X1 travels with the max feed rate  but X2 goes much faster.
All these tests were made in auto parking mode.

esenapaj, 2016-12-06:
By the way, this issue can be closed?

dp90pusika, 2016-12-06:
Yes I think, thank you so much for the help!

esenapaj, 2016-12-06:
It was my pleasure.

closed 2016-12-06 by esenapaj