esenapaj/Marlin #176: My DRV8825 drivers does not work

esenapaj, 2016-12-17:
_@bruce356 commented 3 days ago_

I have configured and find it does not work with my DRV8825 drivers.

I found this post “dp90pusika commented on Aug 16  - And I'm testing https://github.com/esenapaj/Marlin for DRV8825 drivers because its broken right now, but everything else is working, if you have different stepper drivers it could be working for you”

https://github.com/MagoKimbra/MarlinKimbra/issues/116

Regards - bruce

esenapaj, 2016-12-17:
_@esenapaj commented 12 hours ago_

@bruce356
About problem of DRV8825 on my fork, it was already solved.
If you set the MINIMUM_STEPPER_PULSE in Configuration_adv.h to 2, maybe your DRV8825 works.

reference: [problem with drv8825 stepper](https://github.com/esenapaj/Marlin/issues/156#issuecomment-258585202)

esenapaj, 2016-12-17:
_@bruce356 commented 31 minutes ago • edited_

Hi esenapaj, thanks for the information it worked a treat. Sorry to have to bother you hope you do not mind.

I have the RADDS LCD  display also have a “RepRapDiscount FULL GRAPHIC Smart Controller” that works with Radds / Repetier Dev  http://www.thingiverse.com/thing:1740725/#comments

I have activated the “LCD support” with RADDS LCD  display currently installed (I will worry about SD later).

I only get the home screen with no functionality no menus, the reset button works but that’s all. Settings are :-

#define ULTRA_LCD   // Character based
#define INDIVIDUAL_AXIS_HOMING_MENU
#define RADDS_DISPLAY

Hoping you can assist Photo Attached

Regards - Bruce

esenapaj, 2016-12-17:
About LCD support of RADDS on my fork, it's still underdeveloped.
I will confirm that afterward.

bruce356, 2016-12-17:
@esenapaj
I also find the soft end stops do not work. After G28 homing M605 S1, T0 or T1 command and the carriages crash into one another, the speed at which they race across the X axis is insane this is just too damaging.
regards bruce

esenapaj, 2016-12-17:
About RADDS LCD, maybe I've fixed just now.
I've created a test branch, please test it.
https://github.com/esenapaj/Marlin/tree/test-radds-lcd

bruce356, 2016-12-17:
Hi esenapaj, I have moved away from the M605 S1 and am simply using “Full Control Mode”, that’s much better.

The reason the X carriage moved at such high speeds is because I set “#define DEFAULT_MAX_FEEDRATE          { 500, 500, 6, 30 }” the 500mm/sec was a bit fast.

 

Now the display works partially I have the normal First screen but when I click on the Knob to move to the next screen it comes up for a fraction of a second and then automatically returns to the first screen. It does this again and again every time I press the knob and also if I just press and hold down the Knob. Its probably something simple. 

 

I have attached my config. files.

 

Regards - Bruce

 

https://github.com/esenapaj/Marlin/tree/test-radds-lcd

esenapaj, 2016-12-17:
Well, where is the config files?

bruce356, 2016-12-17:
I simply attached them to the email I sent you, what is the easiest way to get them to you

Regards - bruce

bruce356, 2016-12-17:
I put it in Dropbox link below

https://www.dropbox.com/s/3bwfvizpui95kkg/Marlin%20config%20files.zip?dl=0

esenapaj, 2016-12-17:
Any file didn't attached with notification mail from GitHub.
I guess that attached file is invalidated by GitHub mailing system.
And I've downloaded your config files from dropbox just now, thanks.
I'll see it from now.

>what is the easiest way

If you rename from `Configuration.h` and `Configuration_adv.h` to  `Configuration.txt` and `Configuration_adv.txt`,
and dragging & dropping those to comment window in this thread, you can attach those files with your comment.
Or, simply copy and paste the contents of the configuration files as comments.

esenapaj, 2016-12-17:
>when I click on the Knob to move to the next screen it comes up for a fraction of a second and then automatically returns to the first screen.
>It does this again and again every time I press the knob

I confirmed it with your config, and I found a (temporary) solution.
If you enable the `SDSUPPORT`, problem is solved.
But I'm not sure why this happens at the moment.
This strange phenomenon occurs not only with `RADDS_DISPLAY` but also with `REPRAP_DISCOUNT_SMART_CONTROLLER`.

bruce356, 2016-12-17:
Conducted a quick test and yes it works as long as the Return button is not pressed.
The RADDS LCD has a reset button and a return to previous menu button. If the "Return to Previous menu" button is pressed the screen returns to the Info Screen (first screen) and is then stuck on that screen. The only option is then to press the Reset button to reset the display and return it to normal function.
When a SD card is inserted the Info screen tells me "Card Inserted" yet the next screen states "No SD Card" and it sees no files on the SD card, this is not a big deal for me as I use Pi3 with Octopi.
Thanks again and regards - Bruce

esenapaj, 2016-12-17:
About "Return to Previous menu" button (back button), maybe that function in my fork is being broken at this time.
I haven't `RADDS_DISPLAY` now, therefore I'm emulating it by `REPRAP_DISCOUNT_SMART_CONTROLLER`.
But `REPRAP_DISCOUNT_SMART_CONTROLLER` haven't a back button, so I've to prepare a separated push button and connect for emulating.
I'll start to try repairing back function after tomorrow.

>When a SD card is inserted the Info screen tells me "Card Inserted" yet the next screen states "No SD Card" and it sees no files on the SD card

Oh sorry, I misunderstood the LCD SDSS as CS1 (shared with onboard sd slot). The correct LCD SDSS is CS0.
I've updated the test branch, so now SD should be detected correctly.
https://github.com/esenapaj/Marlin/tree/test-radds-lcd

esenapaj, 2016-12-17:
By the way, I have a couple of things to worry about seeing your configuration file.

1.
```cpp
#if ENABLED(PIDTEMPBED)

  //#define PID_BED_DEBUG // Sends debug data to the serial port.

  //24 volt 220 watts Kapton adhesive MM3 - Create bed heater no insulation under heat spreader.
  // from PID Autotune Dec. 2016
      #define  DEFAULT_bedKp 125.63 // <----- It seems that you'd like to use these values,
      #define  DEFAULT_bedKi 12.8
      #define  DEFAULT_bedKd 308.13
	  
  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define  DEFAULT_bedKp 10.00 // <----- but above those values are overwritten by these.
  #define  DEFAULT_bedKi .023 // therefore, these values must be commented out.
  #define  DEFAULT_bedKd 305.4

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define  DEFAULT_bedKp 97.1
  //#define  DEFAULT_bedKi 1.41
  //#define  DEFAULT_bedKd 1675.16

  // FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
#endif // PIDTEMPBED
```

2.
```cpp
// @section homing

// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_125_160 <----- This is an incorrect usage. Even if you change number part of BED_CENTER_AT_0_0, invalid macro is just created.

// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0 // Distance between the nozzle to printbed after homing
```

esenapaj, 2016-12-17:
About soft end stops and crashing carriages, here is some information that seems to be useful?
https://github.com/MarlinFirmware/Marlin/pull/5175#issuecomment-264916175

>・toolheads can move toward their end stops after homing despite the soft endstops enabled, this goes also for Y, so this problem lies maybe somewhere else.

>・after a restart and hemeing all axes, but not taking any moves with X0 or X1 just issuing a tool change makes the new toolhead crash into the other parking toolhead. 
>(this only happens in S0 then not using S command after T command, if sending T0-1 S1 this doesn't happen)

bruce356, 2016-12-18:
The SD card is now functioning and I can see the G-code files on the card.
>The correct LCD SDSS is CS0.
I've updated the test branch, so now SD should be detected correctly.

Just a mistake I missed that, at this stage I am just testing motion etc.  would have probably noticed the problem when heating the Bed, thanks for checking.
> #define  DEFAULT_bedKp 10.00 // <----- but above those values are overwritten by these.

I was not aware it could not be used in this manner, after inserting my numbers luckily I decide not to use it and commented it out but just left the numbers in place, have now set to default.
>//#define BED_CENTER_AT_125_160 <----- This is an incorrect usage. Even if you change number part of BED_CENTER_AT_0_0, invalid macro is just created.

I will stick with full control mode for now until this gets sorted - thanks
>About soft end stops and crashing carriages

Regards - Bruce

bruce356, 2016-12-18:
Hi esenapaj, finally wanted to do a print and increased temperatures to normal extruding (from 25 deg testing to 200 deg PLA). Found the Fans do not work, in S3D I set the bed cooling fans to gradually increase in speed to keep hot end temp. more stable.

 

I can see in the LCD the speed increasing as it should but it’s feeding  the wrong fan (hot end fan), you have the bed fan pin connected to the hot end pin it seems (maybe pin 8 and 9 wrong way round).

 

I think it should be :-

#define ORIG_FAN_PIN            9  for hot end

#define ORIG_FAN2_PIN            8  for bed fan

 

The Hot end fan does not start/turn on at all because it will not start until it reaches somewhere around 235 + and I do not run my bed fans that fast or high.

 

Regards - Bruce

 

 

From: esenapaj [mailto:notifications@github.com] 
Sent: Sunday, 18 December 2016 9:03 AM
To: esenapaj/Marlin
Cc: bruce356; Mention

esenapaj, 2016-12-18:
>S3D

What is the S3D?

>ORIG_FAN_PIN
>ORIG_FAN2_PIN

These are invalid macro on Marlin. Maybe it's a Repetier-Firmware's rule.

And, second fan should be `FAN1_PIN`. not a `FAN2_PIN`.
If you set two fans as follows
```cpp
#define FAN_PIN             9
#define FAN2_PIN            8
```
, it causes iniquitous behavior.
You should set like
```cpp
#define FAN_PIN             9 // <----- first fan
#define FAN1_PIN            8 // <----- second fan
```
.

bruce356, 2016-12-18:
S3D is Simplify3D a Slicer program.
I just included the 
#define FAN_PIN             9
#define FAN2_PIN            8
as an example from MK4DUO
I was hoping you could fix this, I know very little about programing

esenapaj, 2016-12-18:
Well, as I mentioned previous comment,

__second fan should be `FAN1_PIN`. not a `FAN2_PIN`.__

__You should set like__
```cpp
#define FAN_PIN             9 // <----- first fan
#define FAN1_PIN            8 // <----- second fan
```
__.__

The root part of MK4DUO is __different from Marlin__.
The root part of MK4DUO is  __similer to Repetier-Firmware__.
You __can't bring__ MK4DUO's settings to Marlin (and my 32bit fork) __as is__.

esenapaj, 2016-12-18:
By the way about software endstop, I confirmed that it's didn't work even if it was enabled.
It was disabled automatically at all times.
Because ENABLED() macro had been broken with Arduino Due, then it returned false always until now.
So I fixed it in main branch few hours ago. Now  software endstop should be worked.
https://github.com/esenapaj/Marlin/tree/Duo
Therefore you can test auto-park mode with software endstop again (but please be careful).

esenapaj, 2016-12-18:
About pin number of MOSFETs:
[image]

bruce356, 2016-12-18:
Hi esenapaj, I think there is something wrong in the code, with repetier and MK4DUO I have never had to reconfigure the pins to make the Extruder Fans function.
What I have changed to make the fans work:-
Originally
 #define FAN_PIN             9
#define FAN1_PIN            8

Changed to
#define FAN_PIN             8
#define FAN1_PIN            9

Originally (Configuration_adv.h)
#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  

Changed to (Configuration_adv.h)
#define E0_AUTO_FAN_PIN 9
#define E1_AUTO_FAN_PIN 9
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  

Regards - Bruce

esenapaj, 2016-12-18:
>```cpp
>#define E0_AUTO_FAN_PIN 9
>#define E1_AUTO_FAN_PIN 9
>```

Why are there same pin number?

bruce356, 2016-12-18:
@esenapaj, Excelent Job it works now respects soft end stops and changes tools as it should without collisions and RADDS LCD works very nicely also.
>Now software endstop should be worked.

I tested my “RepRapDiscount FULL GRAPHIC Smart Controller” and the initial "Info Screen" comes up, but the screen has a very lite pulsing and when the screen is changed to the next it screws up and becomes unreadable, not very important but it would be nice if both LCD types worked on the RADDS.

For a lot of people the nock off “RepRapDiscount FULL GRAPHIC Smart Controller” would be a lot cheaper.

Regards - Bruce

esenapaj, 2016-12-18:
>it works now respects soft end stops and changes tools as it should without collisions

It's a really good news.

>I tested my “RepRapDiscount FULL GRAPHIC Smart Controller” and the initial "Info Screen" comes up, 
>but the screen has a very lite pulsing and when the screen is changed to the next it screws up and becomes unreadable

Hum...? it's a bit strange.
Because I also have `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, and it works fine with RADDS.
I'd like to more comment but I've to go out now, therefore it's a while later.

bruce356, 2016-12-19:
Hi, I find that any reasonable temperature control does not work, its broken.
When setting temp on hot end to 204 deg. the fluctuation is from 197/198 up to 212/213 degrees.
Does not matter Bang Bang or PID its always the same, change PID settings makes no diff. still the same fluctaution. 

To test hardware I loaded Repetier Dev. and it maintains temperature within 203.9 and 204.1 deg. that is very accurate temperature.

Regards - bruce

esenapaj, 2016-12-20:
About `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`:

| Wiring (reference: [Setup for DUE with RADDS and Graphical LCD](https://github.com/luc-github/DUE-RADDS-GLCD), [LCD for Smoothieboard](http://smoothieware.org/panel)) | Converting cable | Result |
| :-: | :-: | :-: |
| [image] | [image] | [image] |

bruce356, 2016-12-21:
hi esenapaj, thanks for the FULL GRAPHIC SMART CONTROLLER wiring info, as you know I use a small circuit board. I have checked against your schematic and there are minor differences, the circuit board GLCD works with Repetier firmware.

EXP1 to Radds LCD connector:-
EXP1 Pin9 (Gnd.) is not connected to anything, so no connection to Radds LCD connector Pin1 (Gnd.). 
All other connections are as per Schematic. Not sure if this Gnd. is important.

EXP2 to Radds LCD connector:-
EXP2 Pin3 is connected to  Radds LCD connector Pin11 (yours is to Pin12).
EXP2 Pin5 is connected to  Radds LCD connector Pin12 (yours is to Pin11).
So the above are reversed but that should not make any difference to the display.
EXP2 Pin9 is not connected to anything, so no connection to Radds Ext. SD connector Pin5.
EXP2 Pin10 is not connected.
[image]

I have found that with only the usb cable from the laptop connected (5V to Radds) the LCD has a legible screen and I can negotiate through the menus, but as soon as I turn on the mains power (24volts to Radds) the screen becomes a mess and it is impossible to determine or read anything off the screen.

Regards - Bruce

esenapaj, 2016-12-21:
>EXP1 Pin9 (Gnd.) is not connected to anything, so no connection to Radds LCD connector Pin1 (Gnd.).
>EXP2 Pin9 is not connected to anything, so no connection to Radds Ext. SD connector Pin5.

Those Gnd pins of EXP1 and EXP2  are being connected to Radds connectors.
Look at this, EXP1 Pin9 is being connected to Radds LCD connector Pin1,
And EXP2 Pin9 is being connected to Radds Ext. SD connector Pin5.
[image]
Perhaps aren't you wiring those Gnds?

esenapaj, 2016-12-21:
>EXP2 Pin3 is connected to Radds LCD connector Pin11 (yours is to Pin12).
>EXP2 Pin5 is connected to Radds LCD connector Pin12 (yours is to Pin11).

This is related to specification of [smart adapter for RAMPS](http://reprap.org/wiki/RepRapDiscount_Smart_Controller) and Marlin.
In case of Marlin with RAMPS, BTN_EN1 of RRD GLCD is treated as BTN_EN2, and BTN_EN2 is treated as BTN_EN1.
It seems to be inverted, but if we connect these correctly (BTN_EN1 to BTN_EN1, BTN_EN2 to BTN_EN2),
behavior of encoder of RepRapDiscount Smart Controller (ASCII LCD) become inverting.
Therefore I'm obeying the rule of Marlin tentatively.

esenapaj, 2016-12-21:
>but as soon as I turn on the mains power (24volts to Radds) the screen becomes a mess and it is impossible to determine or read anything off the screen.

Perhaps surge current? but it's strange because you said it's working with repetier-firmware.

bruce356, 2016-12-21:
Hi esenapaj, currently using my Radds LCD Display that is working well.

If I ever get the Full Graphic Smart Controller working properly, if the encoder is revered I can swap pins 50 and 52 in pins_RADDS.h .  I will try and connect those earth pins to see if that makes any difference.

 

Have you had a chance to have a look at why Marlin has such wild temperature swings, as I reported earlier.

 

Thanks and Regards - Bruce

esenapaj, 2016-12-21:
>Have you had a chance to have a look at why Marlin has such wild temperature swings, as I reported earlier.

I currently received multiple support requests from you, and its processing isn't in time...

Todo list:
・Survey and try to correct the back button of RADDS LCD
・When `SDSUPPORT` is disabled, click button of encoder malfunction
・Cooling fans are malfunctioning?
・Full Graphic Smart Controller with RADDS2LCD doesn't work on Marlin?
・Temperature control is broken?

esenapaj, 2016-12-21:
>if the encoder is reversed I can swap pins 50 and 52 in pins_RADDS.h .

Other solution available.
If you enable `REVERSE_ENCODER_DIRECTION` in `Configuration.h`, the operation is reversed.

bruce356, 2016-12-21:
Hi, 
1. try to correct the back button of RADDS LCD  --  Yes that would be good

2. When SDSUPPORT is disabled, click button of encoder malfunction  -- not so important

3. Cooling fans are malfunctioning?  --  I rectified this and fans are working as should be.
What I have changed to make the fans work:-
Originally
#define FAN_PIN 9
#define FAN1_PIN 8

Changed to
#define FAN_PIN 8
#define FAN1_PIN 9

4. Full Graphic Smart Controller with RADDS2LCD doesn't work on Marlin?  --  Have to try connecting earths to test.
5. Temperature control is broken?  --  **Yes this would be MOST important** one to rectify others can wait.

Regards - Bruce

esenapaj, 2016-12-21:
>wild temperature swings
I'm confirming it...
When I set target temperature to 200℃, fluctuation is from 196 to 203℃. 
[image]

bruce356, 2016-12-22:
I have been doing a little experimenting in my limited capacity.
Made a minor change in "Temperature.h" file:-
 #if ENABLED(PIDTEMP) || ENABLED(PIDTEMPBED)
      #ifdef ARDUINO_ARCH_SAM
	 #define PID_dT (((OVERSAMPLENR + 2) * 14.0) / TEMP_TIMER_FREQUENCY)//12 deg temp fluctuation
     // #define PID_dT (((OVERSAMPLENR + 2) * 14.0)/(F_CPU / 64.0 / 256.0)) //0.4 deg temp fluctuation
      #else
        #define PID_dT ((OVERSAMPLENR * 12.0)/(F_CPU / 64.0 / 256.0))
      #endif

[image]
LEFT hand side of above curve using:-
#define PID_dT (((OVERSAMPLENR + 2) * 14.0)/(F_CPU / 64.0 / 256.0)) //0.4 deg temp fluctuation

RIGHT hand side of above curve using;-
#define PID_dT (((OVERSAMPLENR + 2) * 14.0) / TEMP_TIMER_FREQUENCY)//12 deg temp fluctuation

Regards - Bruce

bruce356, 2016-12-22:
Changed the formula to:-
#define PID_dT (((OVERSAMPLENR + 2) * 14.0)/(F_CPU / 128.0 / 256.0))
this is even better.
Regards - Bruce

esenapaj, 2016-12-22:
I thought that PID autotune (`m303`) solves waving temperature, but it was useless unfortunately.
But PID and ADC section is black box for me, and PID tuning is very influential by individual equipment and environment,
therefore unless I'm confident about it, I don't want to change the constants in this part.
Therefore I introduce a new option (but it's a emergency measure), `PID_dT_FACTOR`.
I'd much appreciate your investigation.

test branch:
https://github.com/esenapaj/Marlin/tree/PID-test

In case of Arduino Due system and current my fork,

`TEMP_TIMER_FREQUENCY` = 16000000 / 64 / 256 = 976.5625 (same as original Marlin)
`(F_CPU / 64.0 / 256.0)` = 84000000 / 64 / 256 = 5126.953125
`(F_CPU / 128.0 / 256.0)` = 84000000 / 128 / 256 = 2563.4765625

. It looks like `PID_dT` define sampling period of the PID routine,
so if you use a `(F_CPU / 64.0 / 256.0)`, PID routine works at over 5 times faster than normal.

When you use the `PID_dT_FACTOR`,

`#define PID_dT_FACTOR 1` = (TEMP_TIMER_FREQUENCY * 1) = 976.5625 (same as normal)
`#define PID_dT_FACTOR 2` = (TEMP_TIMER_FREQUENCY * 2) = 1953.125
`#define PID_dT_FACTOR 3` = (TEMP_TIMER_FREQUENCY * 3) = 2929.6875
`#define PID_dT_FACTOR 4` = (TEMP_TIMER_FREQUENCY * 4) = 3906.25

Result with my machine (24V PSU, as same as you):

|  set `PID_dT_FACTOR` to 1 ... awful, set `PID_dT_FACTOR` to 2 ... good, but bit overshoot | set `PID_dT_FACTOR` to 4 ... good, but bit undershoot, set `PID_dT_FACTOR` to 3 ... maybe best of these? |
| :-: | :-: |
[image] | [image]

esenapaj, 2016-12-22:
Ah I forgot, I'm not sure the reasoning but

```cpp
#define PID_FUNCTIONAL_RANGE = (PID_MAX / Kp)
```
seems to be useful for accurate PID control.
<details>
<summary>like this</summary>

```diff
#if ENABLED(PIDTEMP)
  //#define PID_AUTOTUNE_MENU // Add PID Autotune to the LCD "Temperature" menu to run M303 and apply the result.
  //#define PID_DEBUG // Sends debug data to the serial port.
  //#define PID_OPENLOOP 1 // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
  //#define SLOW_PWM_HEATERS // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
  //#define PID_PARAMS_PER_HOTEND // Uses separate PID parameters for each extruder (useful for mismatched extruders)
                                  // Set/get with gcode: M301 E[extruder number, 0-2]
-  #define PID_FUNCTIONAL_RANGE 10 // If the temperature difference between the target temperature and the actual temperature
-                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.
  #define K1 0.95 //smoothing factor within the PID

  // Emergency measure for suppress PID vibration when PID autotune is useless
  // When you set this to 2, sampling period of the PID routine will be 1/2
  //#define PID_dT_FACTOR 1

  // If you are using a pre-configured hotend then you can use one of the value sets by uncommenting it
  // Ultimaker
  #define  DEFAULT_Kp 22.2
  #define  DEFAULT_Ki 1.08
  #define  DEFAULT_Kd 114

  // MakerGear
  //#define  DEFAULT_Kp 7.0
  //#define  DEFAULT_Ki 0.1
  //#define  DEFAULT_Kd 12

  // Mendel Parts V9 on 12V
  //#define  DEFAULT_Kp 63.0
  //#define  DEFAULT_Ki 2.25
  //#define  DEFAULT_Kd 440

+  #define PID_FUNCTIONAL_RANGE (PID_MAX / Kp) // If the temperature difference between the target temperature and the actual temperature
+                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.
+
#endif // PIDTEMP
```
</details>

Reference: [Hotend overshoots](http://forums.reprap.org/read.php?4,369867)

bruce356, 2016-12-23:
Hi, it seems there is a problem with the PID-Test version of Marlin-32.
Pre-set PIDS in Configuration.h
 // Create-DualX
  #define  DEFAULT_Kp 13.35
  #define  DEFAULT_Ki 0.91
  #define  DEFAULT_Kd 48.90

Your New programing changes the loaded PIDS to:-
#define PID_dT_FACTOR 1
 #define  DEFAULT_Kp 13.3
  #define  DEFAULT_Ki 0.44
  #define  DEFAULT_Kd 99.3

#define PID_dT_FACTOR 2
 #define  DEFAULT_Kp 13.3
  #define  DEFAULT_Ki 0.89
  #define  DEFAULT_Kd 49.65

#define PID_dT_FACTOR 3
 #define  DEFAULT_Kp 13.3
  #define  DEFAULT_Ki 1.33
  #define  DEFAULT_Kd 33.1

#define PID_dT_FACTOR 4
 #define  DEFAULT_Kp 13.3
  #define  DEFAULT_Ki 1.78
  #define  DEFAULT_Kd 24.82

I Know I can change these numbers after woulds but not sure if anything else may be changing.

regards - bruce

esenapaj, 2016-12-23:
Are those values after PID autotune (`m303`)?

bruce356, 2016-12-23:
No that is after I load the firmware to the Due, as I say I know I can change them manually (GLCD) or do autopid. If I have certain PID numbers set in Configuration.h the same numbers should transfer to eeprom and not be changed.

esenapaj, 2016-12-23:
It's a specification.
Innner `Kp`, `Ki`, and `Kd` aren't `DEFAULT_Kp`, `DEFAULT_Ki`, and`DEFAULT_Kd`,
and innner `Kp`, `Ki`, and `Kd` aren't `raw_Kp`, `raw_Ki`, and `raw_Kd` that those are used for displaying on console and LCD.

Your previous settings I guess:
```cpp
#define DEFAULT_Kp 13.35
#define DEFAULT_Ki 0.91
#define DEFAULT_Kd 48.90

#define OVERSAMPLENR 16
#define F_CPU 84000000L
#define PID_dT (((OVERSAMPLENR + 2) * 14.0) / (F_CPU / 128.0 / 256.0)
```

converting formula from `DEFAULT_K*` to `K*` in `temperature.cpp`:
```cpp
    float Temperature::Kp = DEFAULT_Kp,
          Temperature::Ki = (DEFAULT_Ki) * (PID_dT),
          Temperature::Kd = (DEFAULT_Kd) / (PID_dT);
    #if ENABLED(PID_EXTRUSION_SCALING)
      float Temperature::Kc = DEFAULT_Kc;
    #endif
```

stored values in EEPROM:
```cpp
Kp = 13.35
Ki = 0.91 * (((16 + 2) * 14.0) / ((84000000L / 128) / 256)) = 0.08945664
Kd = 48.90 / (((16 + 2) * 14.0) / ((84000000L / 128) / 256)) = 497.4365234375
```

New settings:
```cpp
#define PID_dT_FACTOR 1

#define REFERENCE_F_CPU 16000000 // 16MHz MEGA2560
#define REFERENCE_TEMP_TIMER_PRESCALE 64
#define HAL_REFERENCE_TEMP_TIMER_RATE (REFERENCE_F_CPU / REFERENCE_TEMP_TIMER_PRESCALE) // timer0 of MEGA2560: 16000000 / 64 = 250KHz (sharing with advanced extruder)
#define REFERENCE_TEMP_TIMER_FREQUENCY (HAL_REFERENCE_TEMP_TIMER_RATE / 256) // note: timer0 is in mode3 (8bit fast PWM), 976.5625Hz always
#define TEMP_TIMER_FREQUENCY REFERENCE_TEMP_TIMER_FREQUENCY
#define PID_dT (((OVERSAMPLENR + 2) * 12.0) / (TEMP_TIMER_FREQUENCY * PID_dT_FACTOR))
```

Loaded values from EEPROM:
```cpp
Kp = DEFAULT_Kp // isn't loaded from EEPROM
Ki = 0.08945664
Kd = 497.4365234375
```

How Marlin calculate values for showing to you:
```cpp
#define unscalePID_i(i) ( (i) / PID_dT )
#define unscalePID_d(d) ( (d) * PID_dT )
#define PID_PARAM(param, h) Temperature::param

raw_Ki = unscalePID_i(PID_PARAM(Ki, eindex));
raw_Kd = unscalePID_d(PID_PARAM(Kd, eindex));
```

Showen values on LCD (those bit differ from your values, maybe cause is rounding error):
```cpp
PID-P = Kp = 13.35
PID-I = raw_Ki = ((0.08945664) / (((16 + 2) * 12.0) / (((16000000 / 64) / 256) * 1))) = 0.40444444444444444444444444444444
PID-D = raw_Kd = ((497.4365234375) * (((16 + 2) * 12.0) / (((16000000 / 64) / 256) * 1))) = 110.025
```

To put it simply, stored previous values in EEPROM become useless when `PID_dT` is changed.
You have to destruction the stored previous values in EEPROM by `m502` (load default) and `m500` (store to EEPROM), when you change a `PID_dT_FACTOR`.

Sanitar777, 2016-12-23:
Hello!
Thanks for you work.

I have the following configuration:
Arduino DUE + RADDS + RADDS LCD Display 4x20, Power 24v
H-bot/CoreXY
Two Z Steppers
Two Z Endstops at MAX

On the Repetier firmware, everything works, but I don't like it. Wanted to try for Marlin, but found nothing, except your firmware. Tried MK4duo, but there was not able to configure the firmware for Dual Endstops.

Took your firmware, everything in it is correct as you discussed here - it worked.

But when I try to print the found that from time to time the printing stops for a few seconds, then continues to print. And the second trouble - the layers shifted.

Thank you!

bruce356, 2016-12-26:
Hi esenapaj, thanks for the extensive explanation, I am not a technical programming person, but I do know that if I upload firmware with Kp, Ki, Kd, pre-set for example:-
  // Create-DualX
  #define  DEFAULT_Kp 13.02
  #define  DEFAULT_Ki 0.87
  #define  DEFAULT_Kd 48.58
then that is what is loaded into the eeprom.

If I change the equation in Temperature.h from:-
#define PID_dT (((OVERSAMPLENR + 2) * 14.0)/(F_CPU / 128.0 / 256.0))
to
#define PID_dT (((OVERSAMPLENR + 2) * 14.0)/(F_CPU / 64.0 / 128.0))
The above  Kp, Ki, Kd, are still loaded into the eeprom.

With the Pid-Test version that is not the case, the  Ki, Kd, are changed dramatically.
At this stage I would prefer if the Pid-Test version were not integrated into main branch, it is just as easy to change the Temperature.h file and not have any complications.

Just wondering why only the eeprom hot end PIDS can be changed from the LCD, there is no adjustment for the Bed PIDS.

Thanks and regards - bruce

bruce356, 2016-12-26:
Hi, just confirming what Sanitar777 mentioned above in his post re-stopping, while printing, the printer will suddenly stop for 25 - 30 seconds  and then continue on. This happens whether I am printing with Tool-0 or Tool-1. It occurs with the endstop version and your latest main branch firmware.
Regards - Bruce

esenapaj, 2016-12-31:
>At this stage I would prefer if the Pid-Test version were not integrated into main branch, it is just as easy to change the Temperature.h file and not have any complications.

I'm sorry but I integrated it into main branch. Therefore you need to order the command `m502` and `m500`, or edit the above line in `Tenperature.h` directly.

>Just wondering why only the eeprom hot end PIDS can be changed from the LCD, there is no adjustment for the Bed PIDS.

It's the (official) Marlin specification. But that doesn't mean there is something impossible, I think that it's just that implementation hasn't been done.

>when I try to print the found that from time to time the printing stops for a few seconds

>while printing, the printer will suddenly stop for 25 - 30 seconds and then continue on

In case of DRV8825 at 24V, It looks like that something of problem exists.
http://reprap.org/wiki/A4988_vs_DRV8825_Chinese_Stepper_Driver_Boards#24v_Low_Amperage_Operation

Or, it may be something wrong with serial communication (USB) or SPI communication (SD card).

bruce356, 2017-01-01:
hay esenapaj, I have changed all my drivers to RAPS128 and started a longer dual colour print, while printing it will suddenly stop and on the LCD the "TUNE" menu option disappears, when printing resumes the "TUNE" menu option re-appears on the LCD display. There is also still the problem of tool-1 being slower than tool-0, it seems like it might be linked to acceleration but I am not sure. My observation is that in dual colour printing it seems that Tool-1 does the stopping.
As it is this firmware is un-usable.
Regards - Bruce

Blue-Marlin, 2017-01-01:
A disapearing TUNE-menu means a empty planner buffer.
A empty planner buffer likely means a communication problem.

Did you try an other host-program, or a lower baud rate?
If using Repetier Host try changing
[image]
If the duration of the breaks corresponds with this number an 'ok' got lost somewhere.

bruce356, 2017-01-02:
Hi Blue-Marlin, thanks for the information, I am still connecting at 250K but have changed to a shorter better quality USB cable and that seems to have solved the problem of prints stopping and re-starting after some 20 - 30 seconds. I have a Pi3 connected and am running Octopi.

Am using  esenapaj 32 bit version of Marlin, have tried Repetier Dev for dual x carriage but it needs to be seriously de-bugged, also MK4duo in my experience is no good for dualX.
Regards - Bruce

bruce356, 2017-01-14:
To compare my results with esenapaj,s Marlin 32 and Repetier Dev, printing a parts container 49x49x49mm, walls are 0.8mm thick (2 x 0.4mm extrusion thickness with 0.4mm nozzle) printed with WinBo  Petg, same printer, same filament, same G-Code from S3D only difference is the firmware, you judge
[image]
[image]