esenapaj/Marlin #179: SAM LIN_ADVANCED problem

MagoKimbra, 2017-01-06:
Hello, I tried to turn LIN_ADVANCED with the alligator and my delta.
Both with MK4duo that with your have the same effect. Printing a cube 20 20, in the corners the extruder (Titan E3D) is like if he lost steps, seems to make a strong recoil and then again extruded. The noise is very strong, putting a finger on the gear feels just trying to portray, losing the steps and then back to extrude.
Without LIN_ADVANCED naturally mold very well without any problem.
I do not think it's a step in the calculations the problem otherwise it too would with AVR CPU, but something in timer or how it is managed advanced_ISR.
I can not understand what the problem is, I did various tests, but nothing.
But before the timer when there were different one for the stepper and the other for ADVANCED extruder with the problem not gave ...

Thank's

Macgyver001, 2017-01-10:
Hi Magokimbra,

Perhaps you can try to increase stepper pulse timing?
There is some code difference and Lin_advance is more efficient/faster, that also means that step pulses become shorten, also in my situation (external stepper drivers) they became to short --> loosing steps.
In the configuration_adv file there is an option "MINIMUM_STEPPER_PULSE" increasing the number to 1 or 2 u seconds makes the stepper pulse longer and hopefully resolves the problem?

MagoKimbra, 2017-01-10:
Yes i try.. I put 1 or 2 or 4 but the problem is same!!!